personalrobotics/prpy #30: Deprecate `PlanToConfiguration`?

siddhss5, 2015-01-07:
Is there any need to have both `PlanToConfiguration` and `PlanToConfigurations`? Why not just have the latter and pass a single or multiple goals? I know a single goal is a `numpy.ndarray` and multiple goals are a `list` of `numpy.ndarray` but we should be able to deal with that inside the function call.

psigen, 2015-01-07:
My thoughts:
* I'm not sure this is a good idea from a practical usability standpoint.  We have several planners that fundamentally can't do anything useful with more than one goal, and it seems less understandable for them to implement PlanToConfigurations then immediately fail when given multiple goals.  

* It seems easy enough to stub PlanToConfiguration as a placeholder call to PlanToConfigurations with a single goal.

* It seems like this might also make client code less readable or more error prone, as simple planning from a point to another point would now either involve a superfluous array construction or a potentially incorrect inference of whether single or multiple goals were provided.

mkoval, 2015-01-07:
I agree with @psigen. I don't think the downsides are worth it because it's only ~two lines of code to add a `PlanToConfiguration` stub that calls `PlanToConfigurations`.

jeking04, 2015-01-07:
For what its worth, I agree with pras and mike, especially from a code readability/clarity standpoint. In cbirrt ( https://github.com/personalrobotics/prpy/blob/master/src/prpy/planning/cbirrt.py) I had to actually create a third function that was not a PlannignMethod that both PlanToConfiguration and PlanToConfigurations called. Otherwise, I believe calling PlanToConfigurations from PlanToConfiguration sent you incorrectly back through the planning pipeline. Does this seem like an ok/correct way to handle implement this?

mkoval, 2015-01-07:
Unfortunately, we do need the extra method. There are currently some issues with calling a `@PlanningMethod` from another `@PlanningMethod` on the same class. I believe the root issue is that it tries to clone the environment into itself.

It should be relatively easy to fix this, though.

siddhss5, 2015-01-07:
Ok fine :). 
Should we be throwing an exception if someone sends a list of goals to PlanToConfiguration then?

Perhaps this is a separate issue. Where should input error checking go?

jeking04, 2015-01-07:
This is a good point. Its seems there are two options:

1. We don't explicitly worry about input argument validity in the python wrapper side. We rely on the actual underlying planners themselves to do validity checking and throw exceptions accordingly.

2. We explicitly check input arguments on the python wrapper side, and then throw exceptions if they are bad. This will causes the planning pipeline to properly move onto the next planner, or fail back up to the calling code.

Thoughts?

psigen, 2015-01-07:
I think it makes sense to check for obvious incorrect argument
specifications up front, so that we can be consistent about them.  This is
basic stuff such as:

1) The goal is the correct (duck) type and expected size
2) Quick assertions that any necessary robot properties are valid/exist
(For example, does robot.planner.PlanToConfiguration exist if we are using
PlanToIK, which calls it).

This would be useful to catch major errors faster (before any planning
starts) and more consistently.

If we only depend on the error checking of the planners, we are basically
biasing towards spending time in implementations with the worst error
checking, as they are more likely to return broken solutions or waste time
instead of failing.  Also, their C++ exceptions are going to be much harder
to read and trace.

This is one of those situations where having formal interfaces in the
Python language would have been really helpful.

mkoval, 2015-01-21:
I would prefer to do as much of the error checking in C++ as possible. People who use our planners without the PrPy wrapper should still get reasonable error messages. Environment cloning happens before you enter the planning method, so there isn't much of a performance difference.

I agree with @psigen for legacy planners that we inherited (e.g. CBiRRT) that don't have very robust error checking internally.

In any case, I am closing the issue because we've gone pretty far off topic. It might be better to have this discussion in person or on Slack.

closed 2015-01-21 by mkoval