personalrobotics/prpy #31: Rename SnapPlanner to StraightLinePlanner?

siddhss5, 2015-01-07:
I wonder if it might be more descriptive to rename it based on its new functionality. If you wanted to be super cool, you could rename it `GeodesicPlanner` :).

psigen, 2015-01-07:
+1 for GeodesicPlanner!

siddhss5, 2015-03-09:
I'm guessing 'GeodesicPlanner` is not happening @mkoval?

mkoval, 2015-03-09:
I'm fine with either `GeodesicPlanner` or `SnapPlanner`. I wasn't sure if it was worth making a potentially breaking change just for the cosmetic improvement.

If you feel strongly about it, I can make this change and update AdaPy and HerbPy accordingly.

mkoval, 2015-09-13:
I do like the name `GeodesicPlanner` but, unfortunately, there is no way to make this change without breaking backwards compatibility. We should keep it in mind next time we do make a breaking change.

closed 2015-09-13 by mkoval

cdellin, 2015-09-13:
Why not rename it to `GeodesicPlanner`, subclassed by a wrapper called `SnapPlanner` which prints a warning when it's instantiated?

psigen, 2015-09-14:
:+1: to @cdellin's suggestion.  `warnings.DeprecationWarning` ahoy!

mkoval, 2015-09-14:
I'm fine with that if someone sends me a pull request. I don't feel strongly enough about the name to implement it myself. :)

@cdellin @psigen Any takers?