personalrobotics/prpy #35: Use Parabolic trajectory retimer in RetimeTrajectory

siddhss5, 2015-01-18:
We should be running the parabolic trajectory retimer which obeys velocity and acceleration limits. However to do this we need to clean up the trajectory and remove all intermediate waypoints because the parabolic timer stops at EACH waypoint.

https://github.com/personalrobotics/prpy/blob/2898a89e35682d21d6354b137ad9903855fe7805/src/prpy/base/robot.py#L173

mkoval, 2015-01-21:
Like I mentioned in #34, this is in [this branch](https://github.com/personalrobotics/prpy/tree/feature/SmoothingRefactor). I still need to cleanup the trajectory before calling the smoother, which is one of the reasons that I haven't merged it yet.

siddhss5, 2015-01-21:
OK, thanks! I think you mean _retimer_ in the previous comment. I see it's just a passthrough now:
https://github.com/personalrobotics/prpy/blob/feature/SmoothingRefactor/src/prpy/planning/retimer.py

siddhss5, 2015-01-21:
BTW, `ParabolicSmoother` also needs a cleaned up trajectory. Rosen has a terrible implementation of the cleanup here: 
https://github.com/rdiankov/openrave/blob/master/plugins/rplanners/parabolicsmoother.cpp#L153
You can do it without _any_ dot products by just comparing differences.

closed 2015-03-09 by siddhss5