personalrobotics/prpy #43: PlanToNamedConfiguration fails on arm with latest version of prpy

aaronjoh, 2015-02-09:
Issue first showed up running the table_clearing demo code, but I think I've traced it back to the new version of clone.py in prpy. The Cloned() function doesn't seem to get the right environment (clone_env is None). This is not an issue with 0.2.0 of prpy but is in 0.3.0. 

Steps to recreate:
Run:
rosrun herbpy console.py --sim --viewer
...
In [1]: robot.right_arm.PlanToNamedConfiguration('home')
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
/homes/aaronjoh/tc-ros/src/herbpy/scripts/console.py in <module>()
----> 1 robot.right_arm.PlanToNamedConfiguration('home')

/homes/aaronjoh/tc-ros/src/prpy/src/prpy/base/manipulator.pyc in wrapper_method(*args, **kw_args)
     57             @functools.wraps(delegate_method)
     58             def wrapper_method(*args, **kw_args):
---> 59                 return self._PlanWrapper(delegate_method, args, kw_args)
     60 
     61             return wrapper_method

/homes/aaronjoh/tc-ros/src/prpy/src/prpy/base/manipulator.pyc in _PlanWrapper(self, planning_method, args, kw_args)
    145         robot = self.GetRobot()
    146         with Clone(robot.GetEnv()):
--> 147             Cloned(self).SetActive()
    148             cloned_args = copy.copy(kw_args)
    149             cloned_args['execute'] = False

/homes/aaronjoh/tc-ros/src/prpy/src/prpy/clone.py in Cloned(*instances)
    150             clone_instance = Cloned(instance.GetParent()).GetLink(instance.GetName())
    151         elif isinstance(instance, openravepy.Robot.Manipulator):
--> 152             clone_instance = Cloned(instance.GetRobot()).GetManipulator(instance.GetName())
    153         else:
    154             raise CloneException('Unable to clone object of type {0:s}.'.format(type(instance)))

/homes/aaronjoh/tc-ros/src/prpy/src/prpy/clone.py in Cloned(*instances)
    144     for instance in instances:
    145         if isinstance(instance, openravepy.Robot):
--> 146             clone_instance = clone_env.GetRobot(instance.GetName())
    147         elif isinstance(instance, openravepy.KinBody):
    148             clone_instance = clone_env.GetKinBody(instance.GetName())

AttributeError: 'NoneType' object has no attribute 'GetRobot'

psigen, 2015-02-09:
This is most likely a regression from the recent PR for deferred planning.

As part of that PR, the Clone/Cloned system was rewritten to improve
locking.   Not sure what the specific issue is here at first glance though.

mkoval, 2015-02-09:
We believe the regression is that [L65](https://github.com/personalrobotics/prpy/blob/master/src/prpy/clone.py#L65) needs to be moved below L77. As you noted, `cloned_env` defaults to `None` and is only initialized on L76.

@pkv is verifying that this fixes the issue.

aaronjoh, 2015-02-09:
Yes, L65 (with self.clone_env) below L77 seems to work.

psigen, 2015-02-09:
Please take a look at:

https://github.com/personalrobotics/prpy/pull/44

aaronjoh, 2015-02-09:
Ok, your patch fixes both the simple test and the table_clearing test I ran

mkoval, 2015-02-09:
Fixed in #44.

closed 2015-02-09 by mkoval