personalrobotics/prpy #45: Increase granularity of exceptions thrown in PlanTo functions

ablasdel, 2015-02-10:
It would be a nice enhancement if, when the planners fail, if we can get more information about why it failed. 

For example:
Start configuration in collision
End configuration in collision
No plan found between Start/End

This would make it much easier to know how to react to failures in planning.

mkoval, 2015-10-20:
To track our progress on this:

- @mkoval added collision exceptions to `VectorFieldPlanner` in #184
- @psigen added exceptions for precondition violations in #196 
- @lgw903 added specific start and goal collision exceptions to `SnapPlanner` and `GreedyIKPlanner` in #205

mkoval, 2016-09-11:
This has been largely addressed by #184, #196, and #205.

closed 2016-09-11 by mkoval