personalrobotics/prpy #48: Planning futures documentation

mkoval, 2015-02-11:
It would be great to have some documentation on what `defer=True` does internally. E.g.:

- What is the default behavior if I don't explicitly pass an `executor`?
- What do meta-planners that are inherently multithreaded (e.g. `Ranked`) do if `defer=False`?
- `concurrent.futures` vs `trollius.futures.future`?
- Are we using `deferred`s or `futures`?
- Can I use `inlineCallbacks`? Should I?

This is a follow-up from #47.

psigen, 2015-02-18:
Where should we put this documentation?

mkoval, 2015-02-18:
I suggest adding a section to the planning section of `README.md` for now. We really do need to split up that document, but that's something we should do as a separate PR.

mkoval, 2016-09-11:
This is no longer relevant now that we've removed Trollius support in #278.

closed 2016-09-11 by mkoval