personalrobotics/prpy #63: Use of numpy.isclose method

jeking04, 2015-03-06:
Currently, to use the numpy.isclose method we need to install numpy from pip rather than apt. This is a bit cumbersome. Can we remove this dependency?

cdellin, 2015-03-06:
Since the implementation [0] is only 23 lines, maybe we can just maintain our own version and hot-patch it in when we import numpy if it isn't there?

[0] https://github.com/numpy/numpy/blob/v1.9.1/numpy/core/numeric.py#L2238

psigen, 2015-03-06:
Is this an issue in 14.04 as well or only in 12.04?

mkoval, 2015-03-06:
I believe the version of NumPy shipped with 14.04 has is_close.

More importantly, I'm not sure if we want to use is_close in this context
anyway. Aren't the default tolerances much tigher than what we care about?

siddhss5, 2015-03-09:
OK, how is one supposed to programmatically decide if a configuration is close enough to another?
Is it with: `all(abs(a-b) < robot.GetActiveDOFResolution())`?

mkoval, 2015-03-09:
I see two places where `numpy.isclose` is used:

1. In [`ComputeJointVelocityFromTwist`](https://github.com/personalrobotics/prpy/blob/492533c054a6a0daedcb44dc57a41744879f716a/src/prpy/util.py#L469):
```python
dq_bounds = [(0, max) if (numpy.isclose(q_curr[i], q_min[i])) else
             (min, 0) if (numpy.isclose(q_curr[i], q_max[i])) else
             (min, max) for i, (min, max) in enumerate(bounds)]
```

Here, I think it makes sense to use `GetActiveDOFResolution` like you suggested.

2. In [`PlanWorkspacePath`](https://github.com/personalrobotics/prpy/blob/cda56f1b49781744312e5a19f57be2c9aa5a5887/src/prpy/planning/workspace.py#L174):
```python
while not numpy.isclose(t, traj.GetDuration()):
```
Here, I think this should be replaced with `t < traj.GetDuration()`.

@siddhss5 Thoughts?

psigen, 2015-04-09:
@siddhss5, @mkoval: Has this been addressed/fixed already?

I think @mkoval's suggestions make sense, and while I am fine with using `numpy.isclose()` in general, I don't think it's necessary in our codebase right now since there aren't too many cases where an exact comparison is actually what we want.

closed 2015-07-01 by psigen