personalrobotics/prpy #66: Push grasping support

siddhss5, 2015-03-09:
We discussed this at our meeting last Friday. It would be nice if there were some programmatic support to push-grasping. I am speculating a function in `robot.py` perhaps that looks like:
```
def pushGrasp(self, target, distance, endPose=None):
    ```
    Pushes the target object which ends up at endPose
    @param target target kinbody
    @param distance push distance
    @param endPose defaults to grabbing the target before the push
```
This function will [@jeking, @mkoval, please edit]:
1. Transform the `target` to `endPose` and grab it
1. Move the hand straight by `distance`

What do you think?

siddhss5, 2015-03-10:
Thinking more, I wonder if this should be in `prpy/planning` instead implemented as a `PlanningMethod` called `pushGrasp` . I can imagine starting with a kinematic planner and then graduate to quasi-static physics and other fanciness.

mkoval, 2015-04-05:
@jeking04 added this to HerbPy here: https://github.com/personalrobotics/herbpy/pull/15.

closed 2015-04-05 by mkoval