personalrobotics/prpy #67: Workspace planner failure case

jeking04, 2015-03-11:
I believe there is an inconsistency in handling failure of workspace trajectories. Currently in PlanToEndEffectorOffset, a trajectory is created that has three waypoints, the start, a waypint at the desired distance and a waypoint at the max distance. Then the min_waypoint_index passed to PlanWorkspacePath is 1. If PlanWorkspacePath fails, timing is summed for waypoints up to but not including the min_waypoint_index. In this case, this will only be the start point and the plan will incorrectly be marked as a success because min_time will be 0.

I see two possible fixes:

1. Pass min_waypoint_index=2 instead of 1.
2. Change the line that gets waypoints to read:
```
wpts = [traj.GetWaypoints(i) for i in range(min_waypoint_index+1)]
```
This will also require changing two lines above to read:
```
min_waypoint_index=traj.GetNumWaypoints() - 1
```

siddhss5, 2015-03-11:
I agree that's a bug. I like fix 2. Could you quickly make those changes? If not, I'll make them on Monday when I'm back.

jeking04, 2015-03-11:
Yup, no problem. I actually already did it locally, just didn't check it in yet in case I was misunderstanding something.

closed 2015-03-12 by mkoval