personalrobotics/prpy #69: Error when simplifying a trajectory with DOFs that are not a subset of ActiveDOFs

jeking04, 2015-03-12:
Currently if a trajectory contains DOFs that are not currently active, the simplifier can produce a trajectory that is in collision.

mkoval, 2015-04-05:
After implementing #95, I think we should retain this behavior. We explicitly set the DOFs in the trajectory as active in `PostProcessPath`. This is also similar to the behavior of the retiming code in `planningutils`.

If this becomes too inconvenient, we can add a helper function to `util` that sets the DOFs in the trajectory as active.

@jeking04 feel free to re-open this if you disagree. :smile:

psigen, 2015-04-05:
Wait a second, @mkoval can you explain why producing an output that is *in collision* is a valid thing to do here?

I could understand only modifying certain DOFs when simplifying, returning an exception, or returning an unmodified path.

mkoval, 2015-04-05:
The smoother only modifies the active DOFs. It only should collision check the links that are affected by the active DOFs. If other links are in collision in the input trajectory, then it's perfectly reasonable for the output to have the same collisions.

Or do you think that the simplifier is actively pushing those links in collision? I don't know which simplifier this is referring to (`OMPLSimplifier`, `ParabolicSmoother`, `HauserParabolicSmoother`?), so I can't replicate.

closed 2015-08-14 by mkoval