personalrobotics/prpy #71: Update MICORobot to use the new trajectory execution pipeline

mkoval, 2015-03-23:
@jeking04 and I recently merged #64, a major pull request to refactor trajectory shortcutting, smoothing, and retiming. See that pull request for the details.

We need to update the [`MICORobot` class](https://github.com/personalrobotics/prpy/blob/master/src/prpy/base/micorobot.py#L53) to use the new functions. In particular, this splits `ExecuteTrajectory` into `ExecutePath` (for un-timed trajectories) and `ExecuteTrajectory` (for timed trajectories). The [implementations in `Robot`](https://github.com/personalrobotics/prpy/blob/master/src/prpy/base/robot.py#L161) are likely sufficient, so you can entirely remove the overload.

This refactor also uses the PrPy planning pipeline for doing trajectory retiming. [By default](https://github.com/personalrobotics/prpy/blob/master/src/prpy/base/robot.py#L69), it uses OMPL for shortcutting, `HauserParabolicSmoother` for smoothing, and the built-in `ParabolicRetimer` for retiming. You should verify that all three of these work on ADA's continuous joints and, if not, override this in the `MICORobot` subclass.

mkoval, 2015-03-23:
Tagging @Stefanos19 @mklingen @herlant.

Stefanos19, 2015-04-15:
Fixed in https://github.com/personalrobotics/prpy/tree/feature/MICORefactor

closed 2015-04-15 by Stefanos19