personalrobotics/prpy #73: OMPLSimplifier uses timeout instead of timelimit as the planning budget

mkoval, 2015-03-25:
All of our planners use the `timelimit` keyword argument for the planning time budget. The `timeout` keyword is reserved as a timeout on trajectory execution. `OMPLSimplifier` breaks this convention and causes problems when passing `timeout=0` along with `execute=True`. See https://github.com/personalrobotics/herbpy/issues/12.

This raises a deeper issue: if we use `timelimit` on `OMPLSimplifier`, it will no longer be possible to pass separate time limits for planning and path simplification. We typically use a much longer time limit for planning (e.g. 30 seconds) than for simplification (e.g. 5 seconds), so this is undesirable.

The quick fix is to use a different keyword for this time limit, e.g. `smoothing_timelimit`. Alternatively, maybe we should find a more principled method of setting this argument when we refactor the way we call post-processing functions.

@psigen Thoughts?

cdellin, 2015-03-25:
I know I've complained about this before, but I just want to interject here for an outsider's perspective. It doesn't make sense to me that a method called "PlanToXxx()" would ever execute anything (I would expect it to return a trajectory object).  Furthermore, I would expect a parameter named "timelimit" or "timeout" on a "PlanToXxx()" method to affect the planning (not execution).

I would suggest a different set of calls, perhaps "MoveToXxx()", which internally do planning and execution for you.  For these methods, I would expect parameters to be something like "planning_timeout" and/or "execution_timeout".  In the case you raise above, perhaps an explicit "smoothing_timeout" would be useful as you mentioned.

If this is not possible, then *at least* please rename the existing "timeout" parameter to be "execution_timeout" so that it's not extremely confusing.  `</rant>`

mkoval, 2015-03-25:
@cdellin I think that is an orthogonal issue. As you mentioned, would have the same name conflict in `MoveToXxx`. We will still have to settle on a new naming convention.

That being said, I'm open to that suggestion. I also don't like how `PlanToConfiguration` behaves completely differently depending upon whether `execute=True` is passed. We can't quite use your suggestion (replacing `Plan` with `Move`) because some planning methods, e.g. `OptimizeTrajectory` do not start with `Plan`. However, we could do something similar by appending a suffix to the method name; e.g. `PlanToXxxAndExecute`.

cdellin, 2015-03-25:
@mkoval The main thrust of my complaint is that when I read a method call like:

    robot.PlanToHome(timeout=10.0, numretries=3, inparallel=True)

those parameters are clearly *planning* parameters.  I would expect that call to time out after 10 seconds of planning, retry planning 3 times on planning failure, and do planning in parallel (or whatever), because the active verb in the method name is `Plan`.  So, primarily, I am objecting to the use of `timeout` as a parameter that affects *execution* (and not planning) in a method called `Plan`, and I'm arguing that, at least, it be renamed `execution_timeout`.

psigen, 2015-03-25:
Also, I would like to point out the additional absurdity that if you do in
fact want to *just* plan, you have to pass `execute=False`.  I am running a
bunch of planning with no execution at all and every call has
`execute=False` at the end.  It looks really weird.

mkoval, 2015-03-25:
If it wasn't clear from my last reply, **I agree**. We should move `execute=True` to a separate function (`PlanToXxxAndExecute`) and make the existing method (`PlanToXxx`) *just plan*. In this case, the only timeout parameter is the existing `timelimit` argument that controls the planning time budget.

@cdellin The same argument is holds if make this change. What does `timeout` mean here?:
```
robot.MoveToHome(timeout=10.0, numretries=3, inparallel=True)
```

I'm suggesting we go further and *completely remove* the `timelimit` and `timeout` arguments from this function. It should only accept fully-qualified names like `planning_timelimit`, `simplification_timelimit`, `smoothing_timelimit`, `retiming_timelimit`, and `execution_timeout`. Other keyword arguments will be forwarded to the planner (but not the post-processing stages) like normal.

cdellin, 2015-03-25:
@mkoval Ok, sounds great! Sorry for the confusion (-:

I definitely support fully-qualified names, although `Plan(timeout=5)` seems preferable to `Plan(planning_timeout=5)` (i.e. when the verb is clear).

Regarding `Move(timeout=5)`, I suppose I would expect the timeout to affect both planning and execution taken together.  It's not exactly clear to me how that would be implemented, though (perhaps plan with planning_timeout=5, then only execute the result if it would finish before the limit?).  If instead you wanted to specify a planning or execution timeout, full qualification is the way to go.

psigen, 2015-03-25:
Is it better to take a separate dictionaries of kwargs?  It seems scary to
munge strings to figure out what your new keywords are supposed to be.

```
s_options = { timelimit: 2.0 }
e_options = { speed=0.9 }
robot.MoveToHome(timelimit=10.0, numretries=3, inparallel=True,
simplifier_options=s_options, executor_options=e_options)
```

mkoval, 2015-03-25:
@cdellin It would  still be `Plan(..., timelimit=5)`. The time limit arguments will only be re-mapped in the `Move(...)` function since the meaning of just `timelimit` is ambiguous.

@psigen I don't anticipate any string munging. My assumption is that `Move` will only forward arguments to the `Plan()` call. You're responsible for calling the post-processing functions manually if you want more fine-grained control over their arguments.

I expect `Move()` to look something like this (pseudocode, don't try to run this):

```python
def Move(*args, planning_timelimit=None, simplify_timelimit=None,
         retime_timelimit, execution_timeout=None, **kwargs):
    path = Plan(*args, timelimit=planning_timelimit, **kwargs)
    simplified_path = Simplify(path, timelimit=simplify_timelimit)
    traj = Retime(simplified_path, timelimit=retime_timelimit)
    self.ExecuteTrajectory(traj, timeout=execution_timeout)
    return traj
```

I'm okay with adding additional `kwargs` dictionaries for the post-processing steps as well. However, it does introduce a lot of syntax bloat since we most frequently only change time limits.

A call that looks like this:

```python
Move(goal, planning_timelimit=20, simplify_timelimit=1, retime_timelimit=1, execution_timeout=0)
```
would become
```python
Move(goal, planning_options={'timelimit': 20}, simplify_options={'timelimit': 1},
           retime_options={'timelimit': 1}, execution_options={'timeout': 0})
```
At this point, albeit this is an extreme example, it seems like you may be better off calling the post-processing functions yourself.

cdellin, 2015-03-25:
FWIW I'm happy with any of these options, but actually I'm totally fine with `Move()` doing simple string munging on its args (e.g. passing any parameter like `planning_xxx` as `Plan(xxx)`, and explicitly erroring on any argument without a known prefix.  It would actually make the code simpler, no?

mkoval, 2015-03-25:
@cdellin I don't think the code inside `Move` changes very much in any of these options. String munging would certainly make it easier to pass those arguments by hand. :smile:

But I do have the same gut reaction to string munging arguments as @psigen. Are there any popular Python libraries that do this?

psigen, 2015-03-25:
I'm a little afraid of tying to 'planning_' 'simplifier_' 'executor_' since
we could easily have the same parameter collision between two planners or
have a multistage simplifier at some point.

Another option is to explicitly take a named dict version of arguments:

```
options = {'OMPLSimplifier': {'timeout': 2.0}, 'ParabolicSmoother':
{'timelimit': 1.0}, `Smoother`: {'timelimit': 2.0} }  # We can have default
names like 'smoother' too.
robot.Move(timelimit=1.0, detailed_options=options)
```

mkoval, 2015-03-25:
@psigen I agree, but I think that's a separate issue. The issue you're describing is caused by using a meta-planner, not the `Move()` wrapper. This is also not incompatible with the `_options` solution, since the options dict could include a `detailed_options` key.

@psigen @cdellin What do you think about the three options for this specific problem?:

1. exposing fully-qualified timelimit arguments, e.g. `Move(planning_timelimit=25)`
2. munging variable names into a flat dictionary, e.g. `Move(planning_timelimit=25, planning_retries=3)`
3. accepting arbitrary dictionaries of options, e.g. `Move(planning_options={'timelimit':25, 'retries': 3})`

I'm fine with any of them, but have a slight preference for (1) or (3).

cdellin, 2015-03-25:
@mkoval I'm happy with any of them!

My only problem with (1) is, as you mentioned earlier, that it wouldn't support arbitrary arguments that you might want to pass to the simplifier, retimer, or executor.

In fact, you could support both (2) and (3) simultaneously, although you might change (3) to just `Move(planning={'timelimit':25, 'retries':3})` (they're clearly already options) in case a random planner wants to use `Plan(options=something)`.  But then again, I suppose I understand @mkoval and @psigen's aversion to argument prefix matching (-:

mkoval, 2015-03-26:
@psigen Any final words? :smile:

mkoval, 2015-03-30:
After speaking with @pkv and @cdellin in person, I think we're going with the `dict` arguments.

@jeking04 Could you take a quick look at this discussion and confirm that it won't be too irritating from a demo perspective?

mkoval, 2015-04-09:
This is fixed in #95.

closed 2015-04-09 by mkoval