personalrobotics/prpy #88: Trajectory returned from RetimeTrajectory() is not smooth

Stefanos19, 2015-04-05:
I tried to use the latest retimer.RetimeTrajectory, similarly to https://github.com/personalrobotics/prpy/blob/7bd38cbb4b65fe58383313af38a06e456e9ea1b1/src/prpy/base/robot.py#L194
but the resulting trajectory for Ada appears to stop at intermediate waypoints and it is not smooth.

mkoval, 2015-04-05:
By default, PrPy should use `HasuerParabolicSmoother` (from the `or_parabolicsmoother` package) to smooth trajectories. It falls back on the built-in `ParabolicTrajectoryRetimer`, which stops at every waypoint, if the `HauserParabolicSmoother` fails.

@Stefanos19 Can you check whether the `HauserParabolicSmoother` is failing? If so, check if `HauserParabolicSmoother` is logging any error messages.

It's possible that `HauserParabolicSmoother` does not properly support continuous joints.

Stefanos19, 2015-04-05:
I see. How can I check if the HauserParabolicSmoother has failed? It looks like the output I get from the smoother is:
[ParabolicSmoother.cpp:158 PlanPath] Detected 6 DOFs.
[ParabolicSmoother.cpp:161 PlanPath] Setting velocity and acceleration limits.
[ParabolicSmoother.cpp:169 PlanPath] Setting joint limits.
[ParabolicSmoother.cpp:200 PlanPath] Setting 8 milestones.
[ParabolicSmoother.cpp:217 PlanPath] Creating collision checker with resolution 0.020000.
[ParabolicSmoother.cpp:230 PlanPath] Shortcutting for 25 iterations.
[ParabolicSmoother.cpp:247 PlanPath] Creating output trajectory (duration: 1.206991).
[ParabolicSmoother.cpp:269 PlanPath] Detected 19 critical ramp transition points.

mkoval, 2015-04-05:
That looks pretty reasonable. The `Detected 19 critical ramp transition points.` message is printed while creating the output trajectory, so it does appear that the `HauserParabolicSmoother` returned success. I'm not sure why the trajectory isn't smooth.

@pkv and I noticed the same behavior, a few times. @pkv Did you ever figure out the root cause?

mkoval, 2015-04-25:
This was a bug in `HauserParabolicSmoother` that we fixed in https://github.com/personalrobotics/or_parabolicsmoother/pull/5. The "blending" feature we added in https://github.com/personalrobotics/or_parabolicsmoother/pull/4 should also `or_parabolicsmoother` produce trajectories that do not stop at waypoints.

closed 2015-04-25 by mkoval