personalrobotics/prpy #91: CloneBindings on Ada will cause a memory leak

mkoval, 2015-04-05:
The current [implementation of `CloneBindings`](https://github.com/personalrobotics/prpy/blob/master/src/prpy/base/mico.py#L128) on Ada directly calls `__init__(sim=True)`. As we recently found out on HERB (https://github.com/personalrobotics/prpy/pull/79), this is a bad idea because it can lead to inadvertently instantiating more than you originally intended. In some cases, this can cause a memory leak.

In fact, this **is causing a memory leak now** because a `ServoSimulator` is being created in the cloned environment. The `ServoSimulator` does not, currently, properly clean up after itself in cloned environments (see https://github.com/personalrobotics/prpy/issues/87).

Stefanos19, 2015-04-06:
What does the CloneBindings do? Should I put a pass as in https://github.com/personalrobotics/prpy/blob/master/src/prpy/base/manipulator.py#L39 ?

mkoval, 2015-04-06:
`CloneBindings` is supposed to copy any PrPy-related metadata associated with the robot. This includes `robot.planner`, `robot.tsrlibrary`, `robot.right_arm`, etc. You can read more [in the PrPy documentation](https://github.com/personalrobotics/prpy#cloning-bound-subclasses).

As a rough guideline, `CloneBindings` should:

1. call the `CloneBindings` function of the superclass (in this case, `prpy.base.Robot`)
2. set any fields that are set by default in `__init__`

In some cases you'll need to create new objects (e.g. `robot.right_arm`, an IK solver). In other cases, you can reference the same objects associated with the parent class (e.g. `robot.planner` or `robot.tsrlibrary`). This depends entirely on whether the field is coupled to a single environment or not.

Stefanos19, 2015-04-15:
Fixed in https://github.com/personalrobotics/prpy/pull/97

closed 2015-04-15 by Stefanos19