personalrobotics/prpy #100: The ik solver of cbirrt cannot be set to the default GeneralIK (or to none), once it is set it to NloptIK.

Stefanos19, 2015-04-06:


mkoval, 2015-04-07:
My understanding was that CBiRRT ignored the IK solver and always used GeneralIK internally.

@cdellin @siddhss5 Do you have a better idea of what's going on in there?

mkoval, 2015-04-09:
CBiRRT seems to be working okay with `nlopt_ik` for me, at least for `PlanToEndEffectorOffset`. What symptoms are you seeing?

Stefanos19, 2015-04-09:
In some cases, the planner runs forever, giving the following repeating output:
```
[cbirrt.cpp:639 PlanPath] Sampling Random Config
[cbirrt.cpp:644 PlanPath] Extending TreeA toward random config
[cbirrt.cpp:1450 MakeNodes] In MakeNodes
[GeneralIK.cpp:931 _SolveStopAtLimits] Projection stalled _numitr: 21
[GeneralIK.cpp:971 _SolveStopAtLimits] Projection successfull _numitr: 3
[cbirrt.cpp:1642 MakeNodes] Configdiff is small
[GeneralIK.cpp:931 _SolveStopAtLimits] Projection stalled _numitr: 8
[GeneralIK.cpp:971 _SolveStopAtLimits] Projection successfull _numitr: 3
[cbirrt.cpp:1642 MakeNodes] Configdiff is small
[cbirrt.cpp:1530 MakeNodes] Extension is going past target
[cbirrt.cpp:664 PlanPath] TreeA MakeNodes Has Generated No New Nodes
[cbirrt.cpp:665 PlanPath] Using Node 0 as target
[cbirrt.cpp:673 PlanPath] Extending TreeB toward random config
[cbirrt.cpp:1450 MakeNodes] In MakeNodes
[GeneralIK.cpp:931 _SolveStopAtLimits] Projection stalled _numitr: 22
[GeneralIK.cpp:971 _SolveStopAtLimits] Projection successfull _numitr: 2
[cbirrt.cpp:1642 MakeNodes] Configdiff is small
[cbirrt.cpp:1530 MakeNodes] Extension is going past target
[cbirrt.cpp:688 PlanPath] TreeB Has Not Reached Target
[cbirrt.cpp:639 PlanPath] Sampling Random Config
[cbirrt.cpp:644 PlanPath] Extending TreeA toward random config
[cbirrt.cpp:1450 MakeNodes] In MakeNodes
[GeneralIK.cpp:931 _SolveStopAtLimits] Projection stalled _numitr: 21
[GeneralIK.cpp:971 _SolveStopAtLimits] Projection successfull _numitr: 2
[cbirrt.cpp:1642 MakeNodes] Configdiff is small
[GeneralIK.cpp:931 _SolveStopAtLimits] Projection stalled _numitr: 8
[GeneralIK.cpp:971 _SolveStopAtLimits] Projection successfull _numitr: 2
[cbirrt.cpp:1642 MakeNodes] Configdiff is small
[cbirrt.cpp:1535 MakeNodes] Extension has stalled
[cbirrt.cpp:664 PlanPath] TreeA MakeNodes Has Generated No New Nodes
[cbirrt.cpp:665 PlanPath] Using Node 0 as target
[cbirrt.cpp:673 PlanPath] Extending TreeB toward random config
[cbirrt.cpp:1450 MakeNodes] In MakeNodes
[GeneralIK.cpp:931 _SolveStopAtLimits] Projection stalled _numitr: 27
[GeneralIK.cpp:971 _SolveStopAtLimits] Projection successfull _numitr: 3
[cbirrt.cpp:1642 MakeNodes] Configdiff is small
[cbirrt.cpp:1530 MakeNodes] Extension is going past target
[cbirrt.cpp:688 PlanPath] TreeB Has Not Reached Target
[cbirrt.cpp:639 PlanPath] Sampling Random Config
[cbirrt.cpp:644 PlanPath] Extending TreeA toward random config
[cbirrt.cpp:1450 MakeNodes] In MakeNodes
[GeneralIK.cpp:931 _SolveStopAtLimits] Projection stalled _numitr: 14
[GeneralIK.cpp:971 _SolveStopAtLimits] Projection successfull _numitr: 2
[cbirrt.cpp:1642 MakeNodes] Configdiff is small
[GeneralIK.cpp:931 _SolveStopAtLimits] Projection stalled _numitr: 8
[GeneralIK.cpp:971 _SolveStopAtLimits] Projection successfull _numitr: 2
[cbirrt.cpp:1642 MakeNodes] Configdiff is small
[cbirrt.cpp:1535 MakeNodes] Extension has stalled
[cbirrt.cpp:664 PlanPath] TreeA MakeNodes Has Generated No New Nodes
[cbirrt.cpp:665 PlanPath] Using Node 0 as target
[cbirrt.cpp:673 PlanPath] Extending TreeB toward random config
[cbirrt.cpp:1450 MakeNodes] In MakeNodes
[GeneralIK.cpp:931 _SolveStopAtLimits] Projection stalled _numitr: 22
[GeneralIK.cpp:971 _SolveStopAtLimits] Projection successfull _numitr: 2
[cbirrt.cpp:1642 MakeNodes] Configdiff is small
[cbirrt.cpp:1530 MakeNodes] Extension is going past target
[cbirrt.cpp:688 PlanPath] TreeB Has Not Reached Target
[cbirrt.cpp:639 PlanPath] Sampling Random Config
[cbirrt.cpp:644 PlanPath] Extending TreeA toward random config
[cbirrt.cpp:1450 MakeNodes] In MakeNodes
[GeneralIK.cpp:931 _SolveStopAtLimits] Projection stalled _numitr: 9
[GeneralIK.cpp:971 _SolveStopAtLimits] Projection successfull _numitr: 2
[cbirrt.cpp:1642 MakeNodes] Configdiff is small
[cbirrt.cpp:1530 MakeNodes] Extension is going past target
[cbirrt.cpp:664 PlanPath] TreeA MakeNodes Has Generated No New Nodes
[cbirrt.cpp:665 PlanPath] Using Node 68 as target
[cbirrt.cpp:673 PlanPath] Extending TreeB toward random config
[cbirrt.cpp:1450 MakeNodes] In MakeNodes
[... 125 lines cut ...]
[cbirrt.cpp:688 PlanPath] TreeB Has Not Reached Target
[cbirrt.cpp:639 PlanPath] Sampling Random Config
[cbirrt.cpp:644 PlanPath] Extending TreeA toward random config
[cbirrt.cpp:1450 MakeNodes] In MakeNodes
[GeneralIK.cpp:931 _SolveStopAtLimits] Projection stalled _numitr: 14
[GeneralIK.cpp:971 _SolveStopAtLimits] Projection successfull _numitr: 2
[cbirrt.cpp:1642 MakeNodes] Configdiff is small
[cbirrt.cpp:1535 MakeNodes] Extension has stalled
[cbirrt.cpp:664 PlanPath] TreeA MakeNodes Has Generated No New Nodes
[cbirrt.cpp:665 PlanPath] Using Node 6 as target
[cbirrt.cpp:673 PlanPath] Extending TreeB toward random config
[cbirrt.cpp:1450 MakeNodes] In MakeNodes
[GeneralIK.cpp:931 _SolveStopAtLimits] Projection stalled _numitr: 27
[GeneralIK.cpp:971 _SolveStopAtLimits] Projection successfull _numitr: 3
[cbirrt.cpp:1642 MakeNodes] Configdiff is small
[cbirrt.cpp:1530 MakeNodes] Extension is going past target
[cbirrt.cpp:688 PlanPath] TreeB Has Not Reached Target
[cbirrt.cpp:628 PlanPath] Sampling IK...
[cbirrt.cpp:861 AddRootConfiguration] Attached Solver Found a Solution!
[GeneralIK.cpp:971 _SolveStopAtLimits] Projection successfull _numitr: 2
[GeneralIK.cpp:971 _SolveStopAtLimits] Projection successfull _numitr: 2
[cbirrt.h:68 Print] ID: 84 Parent: -1 1.62934 -0.401717 -0.291346 5.4221 1.24591 0.529376 
[cbirrt.cpp:947 AddRootConfiguration] Constrained IK solution(s) found!
[cbirrt.cpp:639 PlanPath] Sampling Random Config
```

mkoval, 2015-04-09:
@siddhss5 Any ideas?

siddhss5, 2015-04-10:
Hmm, why isn't this using `GeneralIK` again? The logs above seem to suggest it is.
Also, this is worrisome:
https://github.com/personalrobotics/comps/blob/master/cbirrt2/cbirrt.cpp#L311

Stefanos19, 2015-04-10:
Note that if I do not set the iksolver to NloptIK, the cbirrt succeeds.

mkoval, 2015-06-23:
This was resolved while we were working on the demo.

closed 2015-06-23 by mkoval