personalrobotics/prpy #99: VectorField planner fails for Ada

Stefanos19, 2015-04-06:
The planner throws an error: ```Deviated from straight line constraint```

siddhss5, 2015-04-07:
Not sure if this is a bug really. I instrumented the planner and it seems to be trying to follow the vector field as much as it can. I wonder if this is incorrect:
```
self.robot.GetDOFVelocityLimits()
Out[2]: 
array([ 0.87266462,  0.87266462,  0.87266462,  2.  ,  2.  , 2.  ,  0.78539816,  0.78539816])
```
The [spec sheet] (http://kinovarobotics.com/wp-content/uploads/2015/01/Mico-Specification-Sheet-v1.0.8.pdf) seems to suggest that the `large actuators` have a limit of `9 RPM` which is `0.942 rad/s`. and the `small actuators` have a limit of `18 RPM`.

Stefanos19, 2015-04-08:
I see. The reason I am a bit suspicious is that I have not been able to have the planner succeed for Ada, even for small displacements (< 0.03).

psigen, 2015-04-08:
@Stefanos19: Can you make a small test script that just loads Ada into an empty environment, and then fails to plan a 1cm pure straight-line path using just the VectorFieldPlanner? 

@mkoval can show you how to directly call the planner if you need help.

mkoval, 2015-04-08:
I've actually never seen `VectorFieldPlanner` succeed on HERB, either. I did this call in the `home` configuration:

    robot.vectorfield_planner.PlanToEndEffectorOffset(robot, numpy.array([0,0,1.]), 0.01)

It throws this exception:

> PlanningError: Step size too small, unable to progress

I think @jeking04 ran into this issue as well and had to change something about the the `dt_multiplier` parameter to get it to work.

Stefanos19, 2015-04-08:
@psgen: I updated a script that tests  the vectorfield planner https://github.com/personalrobotics/ada_cbirrt_test/blob/master/src/vectorfield_planner_test.py
I get either the `Deviated from straight line constraint` or the ` Step size too small, unable to progress` error.

siddhss5, 2015-04-09:
Looks like some of my _enhancements_ to the planner broke it. Sorry. I think some of these issues should be fixed now. Let me know. I'm at DARPA away from a real computer.

Stefanos19, 2015-04-09:
With the latest changes, it works for displacements up to 5cm, but then it will throw the `deciated from straight line constraint` error.

mkoval, 2015-04-09:
Do any of the other planners work with changes > 5 cm? ADA is only 6-DOF,
so there are no free degrees of freedom while following this type of
constraint. It might simply not be possible to do any better.

siddhss5, 2015-04-09:
Could the poor performance also be related to https://github.com/personalrobotics/or_urdf/pull/11 and https://github.com/personalrobotics/ada/issues/9? Also note that by the nature of its construction, `PlanToEndEffectorOffset` does *not* actively try to prevent deviation from the straight line. It merely keeps moving straight ahead [all the vectors are pointing straight, not inward].

mkoval, 2015-04-09:
I think we should change `PlanToEndEffectorOffset` to make the vector field point toward the straight line constraint. I am seeing the same thing as @Stefanos19 I can sequentially call the planner many times with 1 cm distances and it succeeds, but it almost always fails with larger distances. This could easily be caused by drift from the constraint.

siddhss5, 2015-04-09:
This was legacy from `MoveHandStraight`, where it was important to make sure that the hand faced the object that was being grasped. If you want me to change this, I can do that but let's all pause for a minute and agree this is actually what we want. :)

mkoval, 2015-04-09:
Regarding those issues that you mentioned:

- https://github.com/personalrobotics/ada/issues/9: It turns out that the numbers in the `.robot.xml` file were right, so this is unrelated. I also ported those numbers to the `.urdf`. file.
- https://github.com/personalrobotics/or_urdf/pull/11: @Stefanos19 is having the same problem on the `.kinbody.xml` file, so this is definitely not related.

mkoval, 2015-04-09:
I think we want `PlanToEndEffectorOffset` to `MoveHandStraight` as far as possible. It doesn't matter what's going on under the hood (i.e. what vector field we're following). It seems like drift is an issue (and it was also a major issue in `MKPlanner`), so I think we should switch to a vector field that compensates for drift.

I'm not sure what the best way of implementing this is. I took a crack at it in https://github.com/personalrobotics/prpy/tree/feature/VectorFieldConstraint, but it performs quite poorly. So probably not this way. :smile:

siddhss5, 2015-04-09:
This looks all correct to me, so I don't know why performance is poor. One thing we could do is make the loss function, now linear [and hence constant gain as its gradient], to be more unforgiving of larger deviations, like a quadratic [and hence linear gain as its gradient].

Could you be more specific about what happens? Does it behave like the original when you set the `gain_error` to `0`?

Stefanos19, 2015-04-09:
@mkoval Both the cbirrt and the GreedyIK  succeed for more than 10cm, from what I've tested.

siddhss5, 2015-04-10:
Interesting. Could you save a `GreedyIK` trajectory and a `VectorField` one? It might be instructive to figure out what exactly is going on. It's a 6DOF arm, so there is really a 1-1 mapping from workspace to cspace. So, something weird is happening for sure.

Stefanos19, 2015-04-10:
Sure, I uploaded you an example of each on Slack: https://personalrobotics.slack.com/messages/ada

mkoval, 2015-09-13:
I want to revive this issue because @jeking04 and I have not seen `VectorFieldPlanner` succeed once on the block sorting demo. I dug into this a little more last night and think the culprit is how we compute the integration timestep (`dt_step`).

On initialization we compute the `dt_step` that serves as our integration timestep:
```python
# On initialization:
vlimits = robot.GetDOFVelocityLimits(robot.GetActiveDOFIndices())
dt_step = min(robot.GetActiveDOFResolutions() / vlimits)
dt_step *= dt_multiplier # default dt_multipler = 1.01
```
On the HERB we can substitute `vlimits = 2.5 rad/s`, `dof_resolution = 0.02 rad`, which results in a `dt_step = 0.008 s`. In each timestep, we split the motion `dqout * dt_step` into multiple collision checks. We compute the number of collision checks using:
```python
numsteps = int(math.floor(max(abs(dqout*dt_step/robot.GetActiveDOFResolutions()))))
```
If `numsteps = 0`, then we declare that "step size too small; unable to progress" and return failure. On HERB, `dt_step / dof_resolution = 0.4 s/rad`. This means that `dq_out` must be at least `0.4 rad/s` to take a step.

That is a huge value! As a result, most `VectorFieldPlanner` calls on HERB terminate on the first timestep by raising that exception. I suspect the same thing is happening on ADA.

mkoval, 2015-09-13:
Fundamentally, I think the key issue is that *the integration timestep should be independent of the collision checking resolution.* We should choose:

-  the integration timestep based on the tightness of our constraints, the smoothness of the vector field and the cost of evaluating it
- the collision checking resolution based on the minimum size of obstacles in the workspace and the cost of collision checking

There's no reason to think that these two quantities are related. In fact, from my experience on Robonaut over the summer, I found that I needed to use a **much** smaller timestep for integration than for collision checking to follow a straight line constraint.

Ideally, I think we should run a standard implementation of a black-box integrator (e.g. [`scipy.integrate.ode`](http://docs.scipy.org/doc/scipy/reference/generated/scipy.integrate.ode.html)) and let it intelligently choose the step size. After each step, in the vector field evaluation function, we can check joint limits and the termination condition. If it has moved more than DOF resolution, then we split the step up into one or more collision checks.

@siddhss5 @psigen Thoughts?

cdellin, 2015-09-13:
+1 from me for separating these two, I agree there's no reason they should be related to each other in general.

mkoval, 2015-09-17:
Fixed in #184.

closed 2015-09-17 by mkoval