personalrobotics/prpy #101: Problem with BirrtPlanner

jeking04, 2015-04-07:
I see this error often when planning via Birrt:

[rrt.h:80 _InitPlan] no initial configurations
[rrt.h:354 PlanPath] BirrtPlanner::PlanPath - Error, planner not initialized

It is not consistent, i.e. sometimes we are able to plan just fine. When I manually move the arm to reset we seem to be fairly wrapped up, so it could be because the start configuration is near joint limit.

mkoval, 2015-04-07:
You can try switching back to CBiRRT or OMPL for our default randomized planner. I switched to the OpenRAVE BiRRT based on the input of @siddhss5 and @psigen. They found that the OpenRAVE BiRRT is much faster than OMPL's RRT-Connect.

psigen, 2015-04-07:
I've found that this pair of errors generally means that the planner thinks
the initial configuration is violating a limit or in collision.

mkoval, 2015-04-07:
It could be the usual "padded joint limits" problem, like @jeking04 suggested, then. Maybe it's time to actually solve this properly...