personalrobotics/prpy #106: Add basic unit tests for planners

mkoval, 2015-04-07:
We've been burned several times recently by planners failing for silly reasons (e.g. typos in member variable names). There is now quite a large set of planners in PrPy---we use eight in HERB's default planning pipeline alone---and several different `@PlanningMethods`. It's virtually impossible to manually test all of these combinations before making a commit.

I started writing some very basic unit tests ([in this branch](https://github.com/personalrobotics/prpy/tree/feature/PlanningUnitTests)) to help with this. My initial goal is to just make **some call** to each `@PlanningMethod`. My goal here is to catch typos, not compute benchmarks. In most cases, these are very easy queries in the `wamtest` environment shipped with OpenRAVE.

I'd like to get this minimal level of coverage on the planners in HERB's default planning pipeline. This issue is mostly just a list to track our progress:

- [ ] `SnapPlanner`
    - [ ] `PlanToConfiguration`
    - [ ] `PlanToEndEffectorPose`
- [ ] `VectorFieldPlanner`
    - [ ] `PlanToEndEffectorPose`
    - [ ] `PlanToEndEffectorOffset`
    - [ ] `FollowVectorField`
- [ ] `GreedyIkPlanner`
    - [ ] `PlanToEndEffectorPose`
    - [ ] `PlanToEndEffectorOffset`
    - [ ] `PlanWorkspacePath`
- [ ] `CHOMPPlanner`
    - [ ] `PlanToConfiguration`
    - [ ] `PlanToEndEffectorPose`
    - [ ] `OptimizeTrajectory`
- [ ] `BiRRTPlanner`
    - [ ] `PlanToConfiguration`
    - [ ] `PlanToConfigurations`
- [ ] `CBiRRTPlanner`
    - [ ] `PlanToConfiguration`
    - [ ] `PlanToConfigurations`
    - [ ] `PlanToEndEffectorPose`
    - [ ] `PlanToEndEffectorOffset`
    - [ ] `PlanToTSR`
- [ ] `OMPLSimplifier`
    - [ ] `ShortcutPath`
- [ ] `OpenRAVERetimer`
    - [ ] `RetimeTrajectory`
- [ ] `OpenRAVEAffineRetimer`
    - [ ] `RetimeTrajectory`
- [ ] `ParabolicSmoother`
    - [ ] `RetimeTrajectory`

psigen, 2015-07-01:
This is a ton of work.  Maybe we should make intermediate PRs with partial coverage?  Are some of these already merged?

mkoval, 2015-09-17:
I just merged a bunch of these tests in #169.

closed 2015-10-16 by mkoval