personalrobotics/prpy #112: feature/MICORefactor: PlanToNamedConfiguration using cbirrt planner

Stefanos19, 2015-04-11:
The trajectories from robot.PlanToNameConfiguration using the cbirrt planner are all over the place, although they go to the right goal. Is this a matter of shortcutting?

mkoval, 2015-04-11:
Yes, that is why. I had to disable `OMPLSimplifier`, our shortcutting algorithm, because it does not support circular joints. This is what `self.simplifier = None` does in AdaPy. As a result, we are not doing any geometric shortcutting.

There are two options:

1. Increase the duration that we call `HauserParabolicSmoother` and let this do all the shortcutting.
2. Switch to a shortcutting algorithm that does work on circular joints.

(1) would be the ideal solution, but we're not sure how well `HauserParabolicSmoother` works in practice. It also ignores the fact that joints are circular while smoothing, so it may not do a very good job.

(2) Should be relatively easy if you use one of the default OpenRAVE shortcutting algorithms. I'm not sure which (if any) support circular joints, though. So you'll need to check that first.

mkoval, 2015-04-29:
This should be fixed now that we use `HauserParabolicSmoother` on ADA. Please re-open this if it is still an issue.

closed 2015-04-29 by mkoval