personalrobotics/prpy #114: Iksolver reset to none

Stefanos19, 2015-04-11:
This is likely the same issue as https://github.com/personalrobotics/prpy/issues/111, but it appears also in the master version of prpy. After running [Jen's branch] (https://github.com/personalrobotics/ada_meal_scenario/tree/feature/action_framework) once, the iksolver is reset to none probably in some clone-bindings, and the planners that use an iksolver fail.

mkoval, 2015-04-11:
Mostly a note to self...the problem is that CloneBindings only gets called
the first time an environment is used as a clone target. We need to
manually clear parent pointers in the target environment in prpy.Clone.

Stefanos19, 2015-04-12:
Resolved with https://github.com/personalrobotics/prpy/commits/bugfix/CloneUserData

closed 2015-04-12 by Stefanos19