personalrobotics/prpy #116: manip.PlanToConfiguration in Ada results in program to freeze

Stefanos19, 2015-04-12:
With the latest master versions of PrPy and Adapy (as well as with the tagged versions prpy 0.5.0  and ada 0.1.2), manip.PlanToConfiguration results in the program to freeze. Here is the output:
```
/homes/snikolai/catkin_demo_jen_branch/src/ada_meal_scenario/src/ada_meal_scenario/actions/trajectory_actions.py:36: FutureWarning: comparison to `None` will result in an elementwise object comparison in the future.
  first_config = cspec.ExtractJointValues(first_wpt, robot, manip.GetArmIndices())
/homes/snikolai/catkin_demo_jen_branch/src/prpy/src/prpy/base/manipulator.py:99: FutureWarning: comparison to `None` will result in an elementwise object comparison in the future.
  return self.GetRobot().GetDOFValues(self.GetIndices())
[INFO] [ada_meal_scenario:trajectory_actions.py:41]:_run: Planning to start of trajectory for action LOOKING_AT_PLATE
[environment-core.h:116 Environment] setting openrave home directory to /homes/snikolai/.openrave
[odecollision.h:124 ODECollisionChecker] ode will be slow in multi-threaded environments
[environment-core.h:193 Init] using ode collision checker
[environment-core.h:848 SetCollisionChecker] setting ode collision checker
[qtcoinviewer.cpp:2928 UpdateFromModel] timeout of GetPublishedBodies
[basemanipulation.cpp:117 main] BaseManipulation: using BiRRT planner
[taskmanipulation.cpp:179 main] using BiRRT planner

```

mkoval, 2015-04-12:
It might not be a freeze, just a really long timeout. Did you check the default timeout on the BiRRT planner?

Stefanos19, 2015-04-13:
I see. However setting robot.PlanToConfiguration works great. Should there be a difference between the two?

mkoval, 2015-04-13:
What do you mean by "setting robot.PlanToConfiguration?"

Stefanos19, 2015-04-13:
Instead of calling manip.PlanToConfiguration, I call robot.PlanToConfiguration

mkoval, 2015-04-13:
`manip.PlanToConfiguration` is clones the environment, sets `manip` to active, then calls `robot.PlanToConfiguration`. They should be equivalent to `robot.PlanToConfiguration` as long as the arm DOFs are active.

Stefanos19, 2015-04-13:
Aha! So something could be breaking in the environment cloning?

mkoval, 2015-04-13:
Can you provide an example that replicates the problem? I am hesitant to clear that something is broken in the cloning unless we have definitive proof.

mkoval, 2015-09-13:
I am closing this issue since it is > 6 months old and we have not been able to replicate it. I suspect that it was the exceedingly long default timeout on CBiRRT (~25 seconds).

closed 2015-09-13 by mkoval