personalrobotics/prpy #118: bugfix/MICORefactor Error message from logger after successful planner call

Stefanos19, 2015-04-15:
Traceback (most recent call last):
  File "/usr/lib/python2.7/logging/__init__.py", line 846, in emit
    msg = self.format(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 723, in format
    return fmt.format(record)
  File "/homes/snikolai/catkin_test_structureIO/src/prpy/src/prpy/logger.py", line 48, in format
    message = self._default_formatter.format(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 464, in format
    record.message = record.getMessage()
  File "/usr/lib/python2.7/logging/__init__.py", line 328, in getMessage
    msg = msg % self.args
TypeError: not enough arguments for format string
Logged from file util.py, line 80

mkoval, 2015-04-25:
@Stefanos19 Is this still an issue in `master`?

Stefanos19, 2015-04-27:
It's been fixed, thanks!

closed 2015-04-27 by Stefanos19