personalrobotics/prpy #133: Add timing info to trajectory via tags

jeking04, 2015-05-06:
In the table_clearing demo we record time for planning, post-processing and execution.  Instead of doing this at the demo level, lets capture this data in prpy and add it as a tag to the output paths/trajectories.  Then table_clearing and future demos/benchmarks can just extract the data from trajectories and log it.

psigen, 2015-05-06:
Uhh.. would it be possible to do this at the herbpy level instead?  Or make
it opt-in?  I already have to disable several loggers that spam planning
output I don't need.

mkoval, 2015-06-23:
Closing because this was added in #136.

@psigen This adds trajectory **tags**, not log output. So there is no additional spam.

closed 2015-06-23 by mkoval

psigen, 2015-06-23:
Ok, that's fine, thanks for the clarification.