personalrobotics/prpy #135: TSR and TSRChain have no deserialize() function

mkoval, 2015-05-12:
These classes have `serialize`, but no `deserialize` function. Also, we may want to consider adding a second pair of `serialize` and `deserialize` functions that operate on dictionaries, instead of strings. This is useful for JSON, YAML, and Pickle serialization.

psigen, 2015-05-12:
Note that the serialize function is the crazy one specifically for cbirrt.
I would not recommend using it for anything else. Instead, consider a
different set of methods like to_json/from_json.

mkoval, 2015-06-23:
I added `serialize_dict` and `deserialize_dict` functions to `TSR` and `TSRChain` in #144. These functions operate on Python `dict` objects that can be serialized with `json` or `yaml`.

psigen, 2015-06-23:
Could we rename these `to_dict()/from_dict()`?  The "serialize" terminology
isn't really used anywhere else in python, but "to"/"from" terminology is
used in numpy, pymongo, and other python libraries.  Thanks!

cdellin, 2015-06-23:
+1 for to_dict/from_dict, serialize to me implies converting to a string or binary.  Converting to a dict representation is different.  The dict could then be serialized e.g. with pickle or json or what have you.

psigen, 2015-06-23:
Created an updated interface here:
https://github.com/personalrobotics/prpy/pull/152

@cdellin: does this look good to you?

cdellin, 2015-06-23:
API looks good!

psigen, 2015-08-28:
@mkoval @cdellin: I think this issue is resolved with the latest few TSR refactors, right?

cdellin, 2015-08-28:
@pseigen I think so!

mkoval, 2015-08-28:
Yup! I'm closing this issue.

closed 2015-08-28 by mkoval