personalrobotics/prpy #143: PlanToConfigurations Error in PlanToTSR

rachelholladay, 2015-06-04:
If you call PlanToTSR with the default chunk size, 1, then the planner calls 'PlanToConfiguration' and everything's fine. However if you call PlanToTSR with a chunk size greater then 1, then 'PlanToConfigurations'  is called . This fails with the error 
" AttributeError: Object <prpy.planning.base.Sequence object at 0x321e150> has no attribute 'PlanToConfigurations'."

Now this doesn't make a lot of sense. The delegate planners (gotten by robot.planners) are listed as:
"Fallback(Sequence(Sequence(SnapPlanner, VectorFieldPlanner, GreedyIKPlanner, Trajopt), TSRPlanner, CBiRRT), NamedPlanner)"

CBIRRT has 'PlanToConfigurations' as a capability  (the others appear not to) which means that the planner, when trying 'PlanToConfigurations' should get to CBIRRT's and use it. For some reason this seems as though it's not happening. 

I was calling PlanToTSR with a pointing tsr and while I'm actually fine with the default chunk size now, the possible bug is curious.  

**Offending Code:** 

point_tsr = self.robot.tsrlibrary(obj, 'point', manip=self.robot.right_arm)
self.robot.right_arm.SetActive()
manip = self.robot.GetActiveManipulator()
 self.robot.PlanToTSR(point_tsr, chunk_size=5)

**Traceback:**

AttributeError                            Traceback (most recent call last)
/usr/lib/python2.7/dist-packages/IPython/utils/py3compat.pyc in execfile(fname, *where)
    173             else:
    174                 filename = fname
--> 175             __builtin__.execfile(filename, *where)

/homes/rmh/ros-hydro/src/pointing/src/binaryLearning/point_tsrrank.py in <module>()
    154     weights = SetWeights(weightFileName)
    155 
--> 156     PointTSR(env, robot).pointAt(bottle, weights)
    157 

/homes/rmh/ros-hydro/src/pointing/src/binaryLearning/point_tsrrank.py in pointAt(self, obj, weights)
     77 
     78         manip = self.robot.GetActiveManipulator()
---> 79         self.robot.PlanToTSR(point_tsr, chunk_size=5)
     80         self.robot.right_hand.MoveHand(f1=2.4, f2=0.8, f3=2.4, spread=3.14)
     81 

/homes/rmh/ros-hydro/src/prpy/src/prpy/base/robot.pyc in wrapper_method(*args, **kw_args)
    103             @functools.wraps(delegate_method)
    104             def wrapper_method(*args, **kw_args):
--> 105                 return self._PlanWrapper(delegate_method, args, kw_args)
    106 
    107             return wrapper_method

/homes/rmh/ros-hydro/src/prpy/src/prpy/base/robot.pyc in _PlanWrapper(self, planning_method, args, kw_args)
    549         from ..util import Timer
    550         with Timer() as timer:
--> 551             result = planning_method(self, *args, **kw_args)
    552         SetTrajectoryTags(result, {Tags.PLAN_TIME: timer.get_duration()}, append=True)
    553 

/homes/rmh/ros-hydro/src/prpy/src/prpy/planning/base.pyc in meta_wrapper(*args, **kw_args)
    175                                                    args, kw_args))
    176             else:
--> 177                 return self.plan(method_name, args, kw_args)
    178 
    179         # Grab docstrings from the delegate planners.

/homes/rmh/ros-hydro/src/prpy/src/prpy/planning/base.pyc in plan(self, method, args, kw_args)
    343 
    344                 try:
--> 345                     return plan_fn(*args, **kw_args)
    346                 except UnsupportedPlanningError:
    347                     continue

/homes/rmh/ros-hydro/src/prpy/src/prpy/planning/base.pyc in meta_wrapper(*args, **kw_args)
    175                                                    args, kw_args))
    176             else:
--> 177                 return self.plan(method_name, args, kw_args)
    178 
    179         # Grab docstrings from the delegate planners.

/homes/rmh/ros-hydro/src/prpy/src/prpy/planning/base.pyc in plan(self, method, args, kw_args)
    238 
    239                     with Timer() as timer:
--> 240                         output = planner_method(*args, **kw_args)
    241 
    242                     logger.info('Sequence - Planning succeeded after %.3f'

/homes/rmh/ros-hydro/src/prpy/src/prpy/planning/base.pyc in __call__(self, instance, robot, *args, **kw_args)
     92                 return wrap_future(executor.submit(call_planner))
     93             else:
---> 94                 return call_planner()
     95 
     96     def __get__(self, instance, instancetype):

/homes/rmh/ros-hydro/src/prpy/src/prpy/planning/base.pyc in call_planner()
     72                 try:
     73                     planner_traj = self.func(instance, cloned_robot,
---> 74                                              *args, **kw_args)
     75 
     76                     # Tag the trajectory with the planner and planning method

/homes/rmh/ros-hydro/src/prpy/src/prpy/planning/tsr.pyc in PlanToTSR(self, robot, tsrchains, tsr_timeout, num_attempts, chunk_size, ranker, max_deviation, **kw_args)
    159                 try:
    160                     if ik_set.shape[0] > 1:
--> 161                         traj = delegate_planner.PlanToConfigurations(
    162                             robot, ik_set)
    163                     else:

/homes/rmh/ros-hydro/src/prpy/src/prpy/planning/base.pyc in __getattr__(self, method_name)
    163         if not self.has_planning_method(method_name):
    164             raise AttributeError("Object {:s} has no attribute '{:s}'.".format(
--> 165                                  repr(self), method_name))
    166 
    167         def meta_wrapper(*args, **kw_args):

AttributeError: Object <prpy.planning.base.Sequence object at 0x439b190> has no attribute 'PlanToConfigurations'.

psigen, 2015-06-04:
I know that @mkoval prefers to customize the delegate planners used by TSRPlanner.  Is it possible that TSRPlanner, when it is constructed in `herbpy`, is passed a `delegate_planner` argument that does not include CBiRRT?

mkoval, 2015-06-23:
@psigen is correct. [Here is the default planner](https://github.com/personalrobotics/herbpy/blob/master/src/herbpy/herbrobot.py#L128) used on HERB:
```python
actual_planner = Sequence(
    # First, try the straight-line trajectory.
    self.snap_planner,
    # Then, try a few simple (and fast!) heuristics.
    self.vectorfield_planner,
    self.greedyik_planner,
    # Next, try a trajectory optimizer.
    self.trajopt_planner or self.chomp_planner
)
self.planner = FirstSupported(
    Sequence(
        actual_planner, 
        TSRPlanner(delegate_planner=actual_planner),
        self.cbirrt_planner
    ),
    # Special purpose meta-planner.
    NamedPlanner(delegate_planner=actual_planner),
)
```

`TSRPlanner` only uses `actual_planner` internally. This intentionally does not include CBiRRT because it has native support for TSRs (which, hopefully, will perform better than `TSRPlanner`). We use `TSRPlanner` to run our "cheap" planners, like `SnapPlanner` and `Trajopt`, that do **not** currently have native support for TSRs. We only fall back on CBiRRT if those planners fail.

The problem is that none of the planners in `actual_planner` support `PlanToConfigurations`. There are a few potential fixes for you:

1. Manually construct a `TSRPlanner` that uses CBiRRT internally:

    ```python
    from prpy.planning.cbirrt import CBiRRTPlanner
    from prpy.planning.tsr import TSRPlanner
    planner =TSRPlanner(delegate_planner=CBiRRTPlanner())
    ```

2. Pass a `chunk_size` of one, which will call `PlanToConfiguration` (instead of `PlanToConfgurations`).
3. Locally edit `herb.py` to add `self.cbirrt_planner` to `actual_planner` (I do not recommend this).

mkoval, 2015-06-23:
This is the expected behavior, so I am closing this bug.

closed 2015-06-23 by mkoval

rachelholladay, 2015-06-23:
Okay your reasoning makes sense to me. Thank you for the detailed explanation.  Help me understand why chunk size should be an argument if it really has to be 1? Is this to allow for your potential fix #1?

mkoval, 2015-06-23:
It makes sense to use a larger `chunk_size` if you are using a planner that supports `PlanToConfigurations`, but does not support `PlanToTSR`. For example, OMPL was in that category for a while.