personalrobotics/prpy #160: ExecutePath fails on Ada: 'JointTrajectory' object has no attribute 'GetDescription'

herlant, 2015-07-20:
I get an error when executing a path with Ada. The error goes away when I comment out https://github.com/personalrobotics/prpy/blob/master/src/prpy/base/robot.py#L392.

In the following trace, the `path` being sent was generated by robot.PlanToConfiguration(dofs) for some hand selected values.

@mkoval @jeking04 

```
In [4]: robot.ExecutePath(path)
[INFO] [prpy.base.robot:robot.py:377]:do_execute: Post-processing path with 2 waypoints.
[WARNING] [prpy.base.robot:robot.py:314]:do_postprocess: Post-processing smooth paths is not supported. Using the default post-processing logic; this may significantly change the geometric path.
[INFO] [prpy.planning.base:base.py:236]:plan: Sequence - Calling planner "ParabolicSmoother".
[INFO] [prpy.planning.base:base.py:245]:plan: Sequence - Planning succeeded after 0.201 seconds with "ParabolicSmoother".
[INFO] [prpy.base.robot:robot.py:387]:do_execute: Post-processing took 0.397 seconds and produced a path with 4 waypoints and a duration of 1.961 seconds.
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
/home/lcv/pr_catkin/src/ada/adapy/scripts/console.py in <module>()
----> 1 robot.ExecutePath(path)

/home/lcv/pr_catkin/src/prpy/src/prpy/base/robot.pyc in ExecutePath(self, path, defer, executor, **kwargs)
    402             return wrap_future(executor.submit(do_execute))
    403         elif defer is False:
--> 404             return do_execute()
    405         else:
    406             raise ValueError('Received unexpected value "{:s}" for defer.'.format(str(defer)))

/home/lcv/pr_catkin/src/prpy/src/prpy/base/robot.pyc in do_execute()
    390             with Timer() as timer:
    391                 exec_traj = self.ExecuteTrajectory(traj, defer=False, **kwargs)
--> 392             SetTrajectoryTags(exec_traj, {Tags.EXECUTION_TIME: timer.get_duration()}, append=True)
    393             return exec_traj
    394 

/home/lcv/pr_catkin/src/prpy/src/prpy/util.pyc in SetTrajectoryTags(traj, tags, append)
    312 
    313     if append:
--> 314         all_tags = GetTrajectoryTags(traj)
    315         all_tags.update(tags)
    316     else:

/home/lcv/pr_catkin/src/prpy/src/prpy/util.pyc in GetTrajectoryTags(traj)
    286     import json
    287 
--> 288     description = traj.GetDescription()
    289 
    290     if description == '':

AttributeError: 'JointTrajectory' object has no attribute 'GetDescription'
```

psigen, 2015-07-20:
Is the `exec_traj` returned by the controller a ROS joint trajectory? It's
supposed to be converted to an openrave trajectory somewhere.

mkoval, 2016-09-11:
@herlant Has this been resolved on ADA?