personalrobotics/prpy #170: GreedyIK fails when called from MoveUntilTouch

mkoval, 2015-09-04:
```In [6]: robot.right_arm.MoveUntilTouch(numpy.array([0., 0., -1.]), 0.1)
[INFO] [prpy.planning.base:base.py:236]:plan: Sequence - Calling planner "Sequence(SnapPlanner, VectorFieldPlanner, GreedyIKPlanner, Trajopt)".
[INFO] [prpy.planning.base:base.py:236]:plan: Sequence - Calling planner "VectorFieldPlanner".
[WARNING] [prpy.planning.base:base.py:254]:plan: Planning with VectorFieldPlanner failed: Step size too small, unable to progress
[INFO] [prpy.planning.base:base.py:236]:plan: Sequence - Calling planner "GreedyIKPlanner".
---------------------------------------------------------------------------
openrave_exception                        Traceback (most recent call last)
/home/mkoval/ros-block-sorting/src/herbpy/scripts/console.py in <module>()
----> 1 robot.right_arm.MoveUntilTouch(numpy.array([0., 0., -1.]), 0.1)

/home/mkoval/ros-block-sorting/src/prpy/src/prpy/base/wam.pyc in MoveUntilTouch(manipulator, direction, distance, max_distance, max_force, max_torque, ignore_collisions, **kw_args)
    260                 manipulator.SetActive()
    261                 traj = manipulator.PlanToEndEffectorOffset(direction, distance, max_distance=max_distance,
--> 262                                                            execute=False, **kw_args)
    263 
    264         collided_with_obj = False

/home/mkoval/ros-block-sorting/src/prpy/src/prpy/base/manipulator.pyc in wrapper_method(*args, **kwargs)
     71             @functools.wraps(delegate_method)
     72             def wrapper_method(*args, **kwargs):
---> 73                 return self._PlanWrapper(delegate_method, args, kwargs)
     74             return wrapper_method
     75 

/home/mkoval/ros-block-sorting/src/prpy/src/prpy/base/manipulator.pyc in _PlanWrapper(self, planning_method, args, kw_args)
    169             cloned_args['execute'] = False
    170             cloned_traj = cloned_robot._PlanWrapper(planning_method,
--> 171                                                     args, cloned_args)
    172 
    173             # Strip inactive DOFs from the trajectory.

/home/mkoval/ros-block-sorting/src/prpy/src/prpy/base/robot.pyc in _PlanWrapper(self, planning_method, args, kw_args)
    588         from ..util import Timer
    589         with Timer() as timer:
--> 590             result = planning_method(self, *args, **kw_args)
    591         SetTrajectoryTags(result, {Tags.PLAN_TIME: timer.get_duration()}, append=True)
    592 

/home/mkoval/ros-block-sorting/src/prpy/src/prpy/planning/base.pyc in meta_wrapper(*args, **kw_args)
    176                                                    args, kw_args))
    177             else:
--> 178                 return self.plan(method_name, args, kw_args)
    344 
    345                 try:
--> 346                     return plan_fn(*args, **kw_args)
    347                 except UnsupportedPlanningError:
    348                     continue

/home/mkoval/ros-block-sorting/src/prpy/src/prpy/planning/base.pyc in meta_wrapper(*args, **kw_args)
    176                                                    args, kw_args))
    177             else:
--> 178                 return self.plan(method_name, args, kw_args)
    179 
    180         # Grab docstrings from the delegate planners.

/home/mkoval/ros-block-sorting/src/prpy/src/prpy/planning/base.pyc in plan(self, method, args, kw_args)
    239 
    240                     with Timer() as timer:
--> 241                         output = planner_method(*args, **kw_args)
    242 
    243                     logger.info('Sequence - Planning succeeded after %.3f'

/home/mkoval/ros-block-sorting/src/prpy/src/prpy/planning/base.pyc in meta_wrapper(*args, **kw_args)
    176                                                    args, kw_args))
    177             else:
--> 178                 return self.plan(method_name, args, kw_args)
    179 
    180         # Grab docstrings from the delegate planners.

/home/mkoval/ros-block-sorting/src/prpy/src/prpy/planning/base.pyc in plan(self, method, args, kw_args)
    239 
    240                     with Timer() as timer:
--> 241                         output = planner_method(*args, **kw_args)
    242 
    243                     logger.info('Sequence - Planning succeeded after %.3f'

/home/mkoval/ros-block-sorting/src/prpy/src/prpy/planning/base.pyc in __call__(self, instance, robot, *args, **kw_args)
     93                 return wrap_future(executor.submit(call_planner))
     94             else:
---> 95                 return call_planner()
     96 
     97     def __get__(self, instance, instancetype):

/home/mkoval/ros-block-sorting/src/prpy/src/prpy/planning/base.pyc in call_planner()
     73                 try:
     74                     planner_traj = self.func(instance, cloned_robot,
---> 75                                              *args, **kw_args)
     76 
     77                     # Tag the trajectory with the planner and planning method

/home/mkoval/ros-block-sorting/src/prpy/src/prpy/planning/workspace.pyc in PlanToEndEffectorOffset(self, robot, direction, distance, max_distance, timelimit, **kw_args)
    134                 traj,
    135                 maxvelocities=0.1*numpy.ones(7),
--> 136                 maxaccelerations=0.1*numpy.ones(7)
    137             )
    138 

openrave_exception: openrave (InconsistentConstraints): [virtual OpenRAVE::PlannerStatus TrajectoryRetimer::PlanPath(OpenRAVE::TrajectoryBasePtr):130] lower limit for traj point 2 dof 6 is not followed (-inf < -1.000000000000000e+06)
```

aaronjoh, 2015-09-12:
Confirmed:

```
aaronjoh@johnson:~/sc-ros/src/block_sorting/scripts$ python run_demo.py --perception-sim --viewer rviz
[WARNING] [root:logger.py:71]:initialize_logging: Install termcolor to colorize log messages.
[ WARN] [1442092809.869844249]: Group 'head_hand' is empty.
[kinbody.cpp:1470 SetDOFValues] dof 4 value is not in limits 0.000000e+00<5.235988e-01
[kinbody.cpp:1470 SetDOFValues] dof 15 value is not in limits 0.000000e+00<5.235988e-01
WARNING: QApplication was not created in the main() thread.
[ INFO] [1442092811.815172268]: Stereo is NOT SUPPORTED
[ INFO] [1442092811.815420616]: OpenGl version: 3 (GLSL 1.3).
[EnvironmentDisplay.cpp:156 EnvironmentChangeSlot] Attempted to switch to environment with non-existent ID 0.
Redecting objects
[INFO] [block_sorting:run_demo.py:97]:<module>: Redecting objects
Press enter to continue, r to redetect
The block name b1  of color ( (0.0, 0.0, 255.0) ) belongs in the  blue bin (idx= 1 )

Grabbing block b1
[INFO] [block_sorting:run_demo.py:120]:<module>: Grabbing block b1
[openravepy_robot.cpp:880 GetActiveManipulatorIndex] GetActiveManipulatorIndex is deprecated
[INFO] [prpy.planning.base:base.py:239]:plan: Sequence - Calling planner "TSRPlanner(Sequence(SnapPlanner, VectorFieldPlanner, GreedyIKPlanner, Trajopt))".
[INFO] [prpy.planning.base:base.py:239]:plan: Sequence - Calling planner "SnapPlanner".
[WARNING] [prpy.planning.base:base.py:257]:plan: Planning with SnapPlanner failed: Straight line trajectory is not valid.
[INFO] [prpy.planning.base:base.py:239]:plan: Sequence - Calling planner "Trajopt".
[INFO] [prpy.planning.base:base.py:248]:plan: Sequence - Planning succeeded after 2.827 seconds with "Trajopt".
[INFO] [prpy.planning.tsr:tsr.py:171]:PlanToTSR: Planned to IK solution set 1 of 3.
[INFO] [prpy.planning.base:base.py:248]:plan: Sequence - Planning succeeded after 5.284 seconds with "TSRPlanner(Sequence(SnapPlanner, VectorFieldPlanner, GreedyIKPlanner, Trajopt))".
[INFO] [prpy.base.robot:robot.py:393]:do_execute: Post-processing path with 10 waypoints.
[ParabolicSmoother.cpp:360 PlanPath] Terminated after 2 iterations and 0.624064 seconds.
[INFO] [prpy.base.robot:robot.py:403]:do_execute: Post-processing took 0.906 seconds and produced a path with 30 waypoints and a duration of 4.943 seconds.
0.119999829052
[INFO] [prpy.planning.base:base.py:239]:plan: Sequence - Calling planner "Sequence(SnapPlanner, VectorFieldPlanner, GreedyIKPlanner, Trajopt)".
[INFO] [prpy.planning.base:base.py:239]:plan: Sequence - Calling planner "VectorFieldPlanner".
[WARNING] [prpy.planning.base:base.py:257]:plan: Planning with VectorFieldPlanner failed: Deviated from straight line constraint.
[INFO] [prpy.planning.base:base.py:239]:plan: Sequence - Calling planner "GreedyIKPlanner".
/home/aaronjoh/sc-ros/src/prpy/src/prpy/planning/workspace.py:130: RuntimeWarning: invalid value encountered in multiply
  max_pose[0:3, 3] += max_distance*direction
Traceback (most recent call last):
  File "run_demo.py", line 123, in <module>
    robot.GrabBlock(block, table, manip=manip)
  File "/home/aaronjoh/sc-ros/src/prpy/src/prpy/base/robot.py", line 116, in wrapper_method
    return delegate_method(self, *args, **kw_args)
  File "/home/aaronjoh/sc-ros/src/herbpy/src/herbpy/action/blocks.py", line 43, in GrabBlock
    timelimit=5)
  File "/home/aaronjoh/sc-ros/src/prpy/src/prpy/base/wam.py", line 298, in MoveUntilTouch
    execute=False, **kw_args)
  File "/home/aaronjoh/sc-ros/src/prpy/src/prpy/base/manipulator.py", line 73, in wrapper_method
    return self._PlanWrapper(delegate_method, args, kwargs)
  File "/home/aaronjoh/sc-ros/src/prpy/src/prpy/base/manipulator.py", line 171, in _PlanWrapper
    args, cloned_args)
  File "/home/aaronjoh/sc-ros/src/prpy/src/prpy/base/robot.py", line 596, in _PlanWrapper
    result = planning_method(self, *args, **kw_args)
  File "/home/aaronjoh/sc-ros/src/prpy/src/prpy/planning/base.py", line 180, in meta_wrapper
    return self.plan(method_name, args, kw_args)
  File "/home/aaronjoh/sc-ros/src/prpy/src/prpy/planning/base.py", line 351, in plan
    return plan_fn(*args, **kw_args)
  File "/home/aaronjoh/sc-ros/src/prpy/src/prpy/planning/base.py", line 180, in meta_wrapper
    return self.plan(method_name, args, kw_args)
  File "/home/aaronjoh/sc-ros/src/prpy/src/prpy/planning/base.py", line 244, in plan
    output = planner_method(*args, **kw_args)
  File "/home/aaronjoh/sc-ros/src/prpy/src/prpy/planning/base.py", line 180, in meta_wrapper
    return self.plan(method_name, args, kw_args)
  File "/home/aaronjoh/sc-ros/src/prpy/src/prpy/planning/base.py", line 244, in plan
    output = planner_method(*args, **kw_args)
  File "/home/aaronjoh/sc-ros/src/prpy/src/prpy/planning/base.py", line 95, in __call__
    return call_planner()
  File "/home/aaronjoh/sc-ros/src/prpy/src/prpy/planning/base.py", line 75, in call_planner
    *args, **kw_args)
  File "/home/aaronjoh/sc-ros/src/prpy/src/prpy/planning/workspace.py", line 136, in PlanToEndEffectorOffset
    maxaccelerations=0.1*numpy.ones(7)
openravepy._openravepy_.openravepy_ext.openrave_exception: openrave (InconsistentConstraints): [virtual OpenRAVE::PlannerStatus TrajectoryRetimer::PlanPath(OpenRAVE::TrajectoryBasePtr):130] lower limit for traj point 2 dof 6 is not followed (-inf < -1.000000000000000e+06)
[INFO] [rospy.core:core.py:387]:signal_shutdown: signal_shutdown [atexit]
```

mkoval, 2015-09-12:
In #174 I discovered that `GreedyIKPlanner` fails if `max_distance` is infinite. I added a workaround in that pull request, but did not fix the underlying issue.;

@siddhss5 @psigen Do you have any better ideas about how to fix this?

siddhss5, 2015-09-13:
I understand the desire to input a `max_distance` of infinity to represent "_I want you to go as far as you can_", but the semantic breaks down for any planner that is not an incremental planner from the start configuration. 

For nearly any other type of planner, it will either return failure, as in `GreedyIK` which needs a finite length workspace path to work with, or result in arbitrarily inefficient behavior, as (I'm guessing) in `cBiRRT` where some arbitrary far away goal is selected and the backwards tree is, in effect, useless (except maybe for biasing). I'd say the _correct_ `cBiRRT` behavior ought to be failure and I worry there's some arbitrary hack underneath to return an answer.

You could try to kludge around that by asking bidirectional searchers to go to `min_distance` instead, but that can lead to silly behavior if `min_distance = 0`. You could try to kludge around that by faking them to be incremental by adding small increments to their plan target at a time, but that makes them very inefficient.

In my opinion, the caller of the `MoveUntilTouch` has the _most_ information about the problem, and ought to provide a reasonable `max_distance`. A `max_distance` of infinity is downright lazy and I wouldn't encourage it.

mkoval, 2015-09-13:
Those are really good points. It doesn't make sense to sample uniformly from `[ min_distance, infinity )` because *no such probability distribution exists*. :sweat:

I'll modify `MoveUntilTouch` to make `max_distance` a required parameter. Temporarily, to maintain backwards compatibility, I'll print a warning if the argument is not specified and default to the current upper bound on the reachability of the WAM (2 meters).

siddhss5, 2015-09-13:
> In general, we can be much cleverer about this with our planners than we are right now. 

For example, we can easily ask OMPL to return the forward tree after a failed run and post-process the tree to see if we've gone beyond the minimum distance. I think @jeking and I spoke about this way back.

mkoval, 2015-09-13:
`or_ompl` doesn't support constraints at all right now, so we don't use it for `PlanToEndEffectorOffset`.

It would still be nice to switch to @cdellin's CBiRRT implementation in OMPL, instead of continuing to use the one in CoMPs. @cdellin @jeking04 do you know what the status of that planner is?

PR #174 (2015-09-13): Fixed GreedyIK planner in MoveUntilTouch
This pull request fixes #170. The problem was that `MoveUntilTouch` defaulted to passing infinity for `max_distance`. This is supported by `VectorFieldPlanner` and (surprisingly) CBiRRT, but not `GreedyIK`.

I fixed this by:
- Check for infinite `max_distance` in the planner and throw an error.
- Modified `MoveUntilTouch` to pass a finite `max_distance`.
closed 2015-09-13 by mkoval