personalrobotics/prpy #179: MoveUntilTouch ignores disabled links

jeking04, 2015-09-16:
Currently the MoveUntilTouch function will reenable any disabled links for objects that are in the ignore_collisions list. For example, the following:

```
with prpy.rave.Disabled(table, padding_only=True):
    traj = manip.MoveUntilTouch(..., ignore_collisions=[table])
```
will enable padding internally and stop the movement at the table padding.  I believe the correct thing to do is to use a KinBodyStateSaver with the LinkEnable option and restore state rather than just calling Enable(True).

mkoval, 2015-09-16:
I fixed this in a recent commit to your fork. It now uses a state saver.

jeking04, 2015-09-16:
Hmm...I didn't get the fix. Is it in a different branch? I also fixed it so I'll try to find and resolve.

mkoval, 2015-09-16:
@jeking04 confirmed that this is fixed in #180.

closed 2015-09-16 by mkoval