personalrobotics/prpy #185: Missing function ComputeEnabledAABB() in prpy.util

DavidB-CMU, 2015-09-17:
I can't run the block demo anymore.

Traceback (most recent call last):
  File "/home/dbutterw/catkin_workspaces/block_sorting_ws/src/block_sorting/scripts/run_demo.py", line 45, in <module>
    env, robot = herbpy.initialize(**herbpy_args)
  File "/home/dbutterw/catkin_workspaces/block_sorting_ws/src/herbpy/src/herbpy/herb.py", line 87, in initialize
    prpy.bind_subclass(robot, HERBRobot, **kw_args)
  File "/home/dbutterw/catkin_workspaces/block_sorting_ws/src/prpy/src/prpy/bind.py", line 366, in bind_subclass
    instance.__init__(*args, **kw_args)
  File "/home/dbutterw/catkin_workspaces/block_sorting_ws/src/herbpy/src/herbpy/herbrobot.py", line 173, in __init__
    import herbpy.action
  File "/home/dbutterw/catkin_workspaces/block_sorting_ws/src/herbpy/src/herbpy/action/__init__.py", line 2, in <module>
    from blocks import GrabBlock
  File "/home/dbutterw/catkin_workspaces/block_sorting_ws/src/herbpy/src/herbpy/action/blocks.py", line 4, in <module>
    from prpy.util import ComputeEnabledAABB
ImportError: cannot import name ComputeEnabledAABB
[INFO] [rospy.core:core.py:387]:signal_shutdown: signal_shutdown [atexit]

mkoval, 2015-09-17:
I suspect you're on the wrong branch of prpy. Are you sure you're on the block_sorting branch? I see the function that branch on Github, so it should exist.

DavidB-CMU, 2015-09-17:
Ok yes I found it the manually, my mistake  :-)
Github's "search this repository" function is not very good I guess.

closed 2015-09-17 by DavidB-CMU