personalrobotics/prpy #190: TSR Bw bounds for rotations

jeking04, 2015-09-23:
I would like to setup a TSR that is unbounded in yaw (or roll or pitch). Previously, I would do this by setting the Bw matrix up as follows:

```
Bw[5,:] = [-numpy.pi, numpy.pi]
```
With the new implementation of TSRs, this gets adjusted so that ```Bw[5,:] = [-numpy.pi, -numpy.pi]```, resulting in no variation in the yaw component when sampling from the TSR.

I have been fixing this by setting the Bw matrix up as follows:
```
Bw[5,:] = [-numpy.pi, numpy.pi - 0.0001]
```
Is there a different way I should be specifying these components?

psigen, 2015-09-23:
This is an ongoing bug in TSR bounds:

@siddhss5 and I are trying to figure out how to handle TSR boundaries, which in various locations are now wrapped or unwrapped to `[-pi, pi)` and `[-pi, pi]`

The short term hack is, as you mentioned, to subtract epsilon.

psigen, 2015-09-29:
@jeking04: I think I might have a fix for this here, can you take a look when you get a chance?
https://github.com/personalrobotics/prpy/pull/194

mkoval, 2015-10-09:
This was fixed in #194.

closed 2015-10-09 by mkoval