personalrobotics/prpy #195: TrajectoryAborted exception

jeking04, 2015-09-30:
Current, the TrajectoryAborted exception is raised if ExecuteTrajectory is called with the robot in a configuration that does not match the first waypoint in the trajectory:

https://github.com/personalrobotics/prpy/blob/master/src/prpy/base/robot.py#L453

Previously, the TrajectoryAborted exception was used to indicate that owd aborted the trajectory.  

This issue is two parts:

1. I can't tell if the TrajectoryAborted exception is still raised when owd aborts a trajectory. Can we verify this is still the case?

2. If it is, can we change the exception thrown when the first waypoint doesn't match the configuration?

This will allow us to differentiate between trajectories not successfully executed by owd and trajectories that never get sent to owd bc they are flawed.

psigen, 2015-09-30:
https://github.com/personalrobotics/prpy/pull/196

mkoval, 2015-10-09:
Fixed in #196.

closed 2015-10-09 by mkoval