personalrobotics/prpy #218: CBiRRT fails unit tests on Travis

mkoval, 2015-10-29:
All of our calls to CBiRRT [fail with this error message](https://travis-ci.org/personalrobotics/prpy/builds/88196188#L2747):

```
Traceback (most recent call last):
  File "/usr/lib/python2.7/unittest/case.py", line 331, in run
    testMethod()
  File "/home/travis/workspace/src/prpy/tests/planning/methods/PlanToEndEffectorPose.py", line 42, in test_PlanToEndEffectorPose_StartInCollision_Throws
    self.planner.PlanToEndEffectorPose(self.robot, goal_ik)
  File "/home/travis/workspace/src/prpy/src/prpy/planning/base.py", line 153, in __call__
    return call_planner()
  File "/home/travis/workspace/src/prpy/src/prpy/planning/base.py", line 133, in call_planner
    *args, **kw_args)
  File "/home/travis/workspace/src/prpy/src/prpy/planning/cbirrt.py", line 89, in PlanToEndEffectorPose
    return self.Plan(robot, tsr_chains=[tsr_chain], **kw_args)
  File "/home/travis/workspace/src/prpy/src/prpy/planning/cbirrt.py", line 194, in Plan
    self.env.LoadProblem(self.problem, robot.GetName())
openrave_exception: openrave (Assert): [/usr/include/boost/smart_ptr/shared_ptr.hpp:653] -> typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = OpenRAVE::IkSolverBase; typename boost::detail::sp_member_access<T>::type = OpenRAVE::IkSolverBase*], expr: px != 0
```

I don't think I've ever seen this error message before. @cdellin, any ideas?

cdellin, 2015-10-29:
Hmm, I expect this is happening here [0]. Was GeneralIK built and available as an OpenRAVE plugin on the machine where the test was run?

[0] https://github.com/personalrobotics/comps/blob/master/cbirrt2/cbirrtproblem.cpp#L111

siddhss5, 2015-10-29:
That was going to be my suggestion too. Spot on @cdellin.

mkoval, 2015-10-29:
Good call @cdellin and @siddhss5. CBiRRT did not `<run_depend>` on `GeneralIK`, so it was never being built. I fixed this in https://github.com/personalrobotics/comps/commit/b1763f9155b47758c27474d48148098e1ab81a08.

closed 2015-10-29 by mkoval