personalrobotics/prpy #220: CHOMPPlanner fails to serialize TSRs

mkoval, 2015-10-29:
I believe this is related to the TSR refactor, when the CBiRRT-format TSR serialization code was moved to CBiRRTPlanner.

```Traceback (most recent call last):
  File "/homes/mkoval/herb-ws/src/prpy/tests/planning/methods/PlanToEndEffectorPose.py", line 66, in test_PlanToEndEffectorPose_GoalInCollision_Throws
    self.planner.PlanToEndEffectorPose(self.robot, goal_ik)
  File "/homes/mkoval/herb-ws/src/prpy/src/prpy/planning/base.py", line 153, in __call__
    return call_planner()
  File "/homes/mkoval/herb-ws/src/prpy/src/prpy/planning/base.py", line 133, in call_planner
    *args, **kw_args)
  File "/homes/mkoval/herb-ws/src/prpy/src/prpy/planning/chomp.py", line 266, in PlanToEndEffectorPose
    releasegil=True, **kw_args
  File "/home/mkoval/storage/herb-ws/src/or_cdchomp/pythonsrc/orcdchomp/orcdchomp.py", line 203, in runchomp
    run = create(mod, releasegil=releasegil, **kwargs)
  File "/home/mkoval/storage/herb-ws/src/or_cdchomp/pythonsrc/orcdchomp/orcdchomp.py", line 130, in create
    cmd += ' start_tsr \'%s\'' % start_tsr.serialize()
AttributeError: 'TSR' object has no attribute 'serialize'
```

cdellin, 2015-10-29:
Yes, CHOMP can plan from/to TSRs, but only supports the legacy serialization format, and the Python wrapper expects that tsrs passed have a `serialize()` method which produces such a serialization.

mkoval, 2015-11-12:
This is a duplicate of #164.

closed 2015-11-12 by mkoval