personalrobotics/prpy #241: Modulo bug in VanDerCorputSampleGenerator

ClintLiddick, 2015-12-08:
https://github.com/personalrobotics/prpy/blob/master/src/prpy/util.py#L1124

`decimals = (s % 1)` will always be `0`, what is this line supposed to mean?

cdellin, 2015-12-08:
I think it's for:
```
>>> 3.5 % 1
0.5
```

DavidB-CMU, 2015-12-08:
@ClintLiddick Did you get an error message?  Please paste in any program output if you see a bug.

Modulo 1 in python will return the bit after the decimal point.

Try running this code:
````
print 0.3 % 1
print 0.5 % 1
print 0.75 % 1
print 1.3 % 1
print 1.75 % 1
````

The output is:
```
0.3
0.5
0.75
0.30000000000000004
0.75
```

This line:
```
decimals = (s % 1)
```
takes the output of the Van der Corput sequence (stored in s) and extracts the bit after the decimal point. 
If it's below half, we increment down, otherwise we increment up. 

If you are now wondering what the larger section of code is doing, I can explain that also. But briefly, it's because we need steps of 1 (actually steps of time t=2), but the binary Van der Corput sequence works with fractions of powers of two.

ClintLiddick, 2015-12-10:
No problem. I was too deep in C++ and couldn't fathom what `x % 1` could possibly mean. No issue.

closed 2015-12-10 by ClintLiddick