personalrobotics/prpy #245: GreedyIK does unnecessary work to generate the error message

mkoval, 2015-12-11:
While reviewing #240 I noticed that the GreedyIK planner generates `collision_error` anytime a call to `FindIkSolution` fails. This requires the expensive process of looping over all IK solutions and calling `CheckCollision` and `CheckSelfCollision` on them.

However, this isn't a terminal error. This means that we may repeat this process several times - even if the final plan is successful.

@lgw903 Can you take a look at this?

DavidB-CMU, 2016-01-07:
@mkoval @lgw903 I am interested to join the discussion on this.  I've noticed some similar code in PlanToEndEffectorPose() in SnapPlanner.

mkoval, 2016-01-09:
I took a look at `PlanToEndEffectorPose` in `SnapPlanner` and don't see an issue. It only calls `FindIkSolutions` to generate an error message if `FindIkSolution` fails. What do you see as the issue?

closed 2016-01-09 by mkoval