personalrobotics/prpy #248: HauserParabolicSmoother ignores timelimit argument on construction

cdellin, 2015-12-15:
I tried to pass the `timelimit` parameter to the `prpy.planning.retimer.HauserParabolicSmoother` constructor, but when I attempt to smooth trajectories, the default limit of 3 sec is used instead.  It appears that the constructor parameter is inserted into a `self.default_options` dictionary which appears to be ignored when the instance's `RetimeTrajectory` planning method is called.  I'm happy to take a stab at this later, but I figured someone who knows this code better would know the short path to a fix.

psigen, 2016-03-27:
@cdellin does #277 cover this?

mkoval, 2016-03-27:
#277 should cover this. @cdellin please re-open this issue if it does not.

closed 2016-03-27 by mkoval