personalrobotics/prpy #256: Print warning if (attempting) to double clone an environment / Refactor workspace.py

DavidB-CMU, 2016-01-14:
This is related to pull request https://github.com/personalrobotics/prpy/pull/55 ("Fixed a number of bugs related to workspace planner") 

If you call a planning method like PlanToEndEffectorOffset() it then calls PlanWorkspacePath(). 
This results in two calls to Clone(), but there's a check in there
```python
if self.clone_env != self.clone_parent:
    with self.clone_parent:
        self.clone_env.Clone(self.clone_parent, self.options)
```
which means that the environment is not cloned twice.

In https://github.com/personalrobotics/prpy/pull/55 @mkoval suggested that we should print a warning message like "Warning: you are trying to clone an environment into itself"
Should we add this warning message? 

I would like to suggest that we also re-factor `workspace.py` so Clone() is not called twice.

psigen, 2016-03-27:
After doing some work on this in #279, I think the best option here is to simply make cloning into yourself be a near-free operation.

There are a lot of cases, like nested `@PlanningMethod` calls, where deduplicating cloning is a very verbose operation, either requiring refactoring tons of common code into private helper methods, or branching off a special case when the clone is not required.

Since `with Clone()` also performs locking and adds a `.Cloned()` operator which normalized references which may or may not be in the environment, it already handles a lot of edge cases for us when we use it.  I think cloning into the same environment should be one of those cases, so we should make it so that there is no penalty for doing so.

Semantically, we will just be accessing/locking the one environment we would be using anyway, so I think this is a safe thing to do.

mkoval, 2016-03-27:
:+1: I agree with @psigen analysis. Eliminating all nested `@PlanningMethod` introduces too much boilerplate to be a viable solution.

I suspect that #279 eliminates most overhead. I am going to close this issue, but please do not hesitate to re-open it if you notice performance issues from multiple `Clone` calls into the same environment.

closed 2016-03-27 by mkoval