personalrobotics/prpy #283: 'Environment' object has no attribute 'Cloned'

DavidB-CMU, 2016-03-26:
The grasp fuze bottle example in HERBpy is broken due to recent changes in prpy clone.py

> Traceback (most recent call last):
>   File "/home/USER/catkin_workspaces/herb_ws/src/herbpy/examples/graspFuzeBottle.py", line 87, in <module>
>     robot.right_arm.PushGrasp(fuze, push_required=False, preshape=grasp_vals)
>   File "/home/USER/catkin_workspaces/herb_ws/src/prpy/src/prpy/base/manipulator.py", line 82, in wrapper_method
>     return delegate_method(robot, obj, manip=self, *args, **kwargs)
>   File "/home/USER/catkin_workspaces/herb_ws/src/herbpy/src/herbpy/action/grasping.py", line 50, in PushGrasp
>     tsrlist=tsrlist, render=render,**kw_args)
>   File "/home/USER/catkin_workspaces/herb_ws/src/herbpy/src/herbpy/action/grasping.py", line 126, in HerbGrasp
>     **kw_args)
>   File "/home/USER/catkin_workspaces/herb_ws/src/prpy/src/prpy/base/manipulator.py", line 73, in wrapper_method
>     return self._PlanWrapper(delegate_method, args, kwargs)
>   File "/home/USER/catkin_workspaces/herb_ws/src/prpy/src/prpy/base/manipulator.py", line 165, in _PlanWrapper
>     with Clone(robot.GetEnv()) as cloned_env:
>   File "/home/USER/catkin_workspaces/herb_ws/src/prpy/src/prpy/clone.py", line 111, in __init__
>     cloned_robot = self.clone_env.Cloned(robot)
> AttributeError: 'Environment' object has no attribute 'Cloned'
> [INFO] [rospy.core:core.py:390]:signal_shutdown: signal_shutdown [atexit]

psigen, 2016-03-27:
Oops, can you check if #284 addresses this?

mkoval, 2016-03-27:
I merged #284. Try the latest `master`.

DavidB-CMU, 2016-03-27:
Yes that fixed it.

closed 2016-03-27 by DavidB-CMU