personalrobotics/prpy #307: `prpy` is unusable without `ros_control_client_py`

psigen, 2016-05-13:
The latest `master` of prpy cannot be imported at all if `ros_control_client_py` is not installed.
```
import prpy
-> import controllers
-> from rewd_controllers import OrController, RewdOrController, RewdOrTrajectoryController
-> from ros_control_client_py import FollowJointTrajectoryClient, TrajectoryExecutionFailed
```

Is it really necessary to put this in `prpy/__init__.py`?
I feel like it is reasonable for people to selectively import controllers when they need them.

mkoval, 2016-05-13:
@ClintLiddick agree that we should definitely make this an optional dependency. Someone in lab got burned by the same issue, so this is high on @ClintLiddick's priority list.

mkoval, 2016-07-21:
Fixed by #308.

closed 2016-07-21 by mkoval